ekstrand/ESP8266wifi #6: HTTP request, returns 'garbage'.

hertog6, 2015-05-03:
I do not know, if it's an serious issue (or it's just my fault ;) ). But I am having some "trouble" with this Arduino library.

Currently I am using the ESP8266 as connection method to a local server, running Apache. The HTTP response I get back from my determines if a connected LED is turned off or on. Based on the "true" or "false" HTTP response. 

The problem I am having is that the HTTP response I am getting isn't complete and garbled. The code I am using to send the HTTP request is as followed:
```C
  String location = "/home-automation/status.php?action=get&light=kitchen"; // Location where to fetch the new Status
  String cmd = "GET " + location + " HTTP/1.1\r\n"; // complete HTTP request
  cmd += "Host: 10.13.37.153\n";

  wifi.connectToServer("10.13.37.153", "80");
  if (wifi.isConnectedToServer())
  {
    Serial.println("I am connected to the server");
    
    wifi.send(SERVER, cmd);

    WifiMessage in = wifi.listenForIncomingMessage(2000);
    if (in.hasData)
    {
      Serial.print("Response: ");
      Serial.println(in.message);
   }

```

I already changed the msgIn[] value to a value of 500, that does increase the amount of char's I get back, but still doesn't un-garble the return data.

OUTPUT with debugging
```
IPD,HTTP/1.1 ��2����0����0��� OK
Date: Sun, 03 May 2015 16:18:59 GMT
Server: Apache/2.4.10 (Debian)
Content-Length: 5
Content-Type: text/plain;charset=UTF-8

false
```

Output without debugging
```
HTTP�/1.1� 20�0 OK�
Da�te: �Sun�, 03� May� 20�15 1�6:27�:09 �GMT�
Se�rver�: Ap�ach�e/2.�4.10� (D�ebian�)
�Cont�ent-�Leng�th:� 5
�Cont�ent�-Typ�e: t�
```

Expected output
```
HTTP/1.1 200 OK
Date: Sun, 03 May 2015 16:18:59 GMT
Server: Apache/2.4.10 (Debian)
Content-Length: 5
Content-Type: text/plain;charset=UTF-8

false
```

ekstrand, 2015-05-03:
Hi Sander,
I my self don't use http and utf-8 encoding as in your case. A tip could be to ask (in the http request) apache to respond in ascii only. 

If that does not help it could be an hardware thing ie esp8266 3.3v and arduino maybe 5v?

God luck!
/ Jonas

hertog6, 2015-05-07:
I haven't tried the "Accept-Charset" request field yet, so I am gonna try that. 

But the ESP8266 is running on main voltage supply is 3.3v and the communication to the RX pin on the ESP8266 is being scaled down to 3.3v due to using an 74HC4050.

ekstrand, 2015-05-08:
I hope Accept-Charset works. A tip is to use an app like SocketTest http://sockettest.sourceforge.net <http://sockettest.sourceforge.net/> for debugging network issues. It lets you set up a tcp server and interact with the client.

I also used logic levels converters when I fist got my ESP8266, but now days I connect the esp and 5v Arduino directly. The logic levels are not harmful as long as the ESP8266 gets a 3.3v supply. Try it on your own risk of course :)

God luck!

ocb00999, 2015-05-15:
I tried to send a POST request, but it is not successful.
Is that supported in this library?

ekstrand, 2015-05-15:
How was it unsuccessful? Can you post the request and answer? Otherwise I cannot help you.
My guess is that it’s charset related. If your web server responds in UTF-8 or some other charset than ASCII-7 you’ll need to find some utf-8 lib for arduino..

/ Jonas

ocb00999, 2015-05-15:
The post request is somthing like this 
POST /sensor HTTP/1.1
X-Parse-REST-API-Key: supersecretapikey
Content-Type: application/json
Host: homesensor.parseapp.com
Content-Length: 40

{"identifier":"dishwasher","state":true}

And for the arduino code I sent is lik this
cmd = "POST /sensor HTTP/1.1 \r\n";
cmd += "X-Parse-REST-API-Key: supersecretapikey\r\n";
cmd += "Content-Type: appliation/json\r\n";
cmd += "Host: homesensor.parseapp.com\r\n";
cmd += "Content-Length: 40\r\n\r\n";
cmd += "{"identifier":"dishwasher","state":true}\r\n";
wifi.send(SERVER,cmd)

By the way, is your library based on AT command?

ekstrand, 2015-05-15:
Hi,
I think I know the problem is. By default the output buffer is limited to 25 characters. Your payload is a lot bigger :)

You’ll have to modify ESP8266wifi.h, see 

Advanced configuration on https://github.com/ekstrand/ESP8266wifi <https://github.com/ekstrand/ESP8266wifi>

And yes, It’s based on AT commands :)

closed 2015-05-30 by ekstrand