ekstrand/ESP8266wifi #7: Missing method for local IP

rpseng, 2015-05-21:
Dear ekstrand,

First of all nice work with this library!

Other libraries have a function called ESP8266Wifi.localIP which returns the local IP. Is there a similar function in you library?

All the best.

ekstrand, 2015-05-21:
I have not needed one yet. If you would like to write one I would be glad to put it in! Have a lot on my plate right now soo it could otherwise be a while until I get the time to write it myself.
/Jonas

rpseng, 2015-05-21:
Hi Jonas, thanks for the prompt reply. I don't actually need it either, just something nice to check if the module is working properly.

I

ekstrand, 2015-05-22:
Ok, you could use the isconnectedtoap function for exactly that. It checks that the module has aquired a valid ip and not 0.0.0.0
:) Jonas

closed 2015-05-30 by ekstrand