ekstrand/ESP8266wifi #8: Question: HTTP wrappers

romangromov, 2015-06-20:
Hi @ekstrand, thanks for your great library.

Is there any known wrappers to support `HTTP` stuff over `ESP8266wifi` library? I mean it would be great to use such a functions like below

```
HttpRequest req = new HttpRequest();
req.setUrl("hello/world");
req.setHeader("Accpet-Encodings", "gzip")
wifi.send(SERVER, req);
```
instead of low-level `Strings` manipulation to build http request message manually.

Thank you in advance!

ekstrand, 2015-06-20:
Hi,
I don't know. If some one wrote one I would be happy to include it in the library. I personally don't use http :)

Hälsningar/Regards
Jonas Ekstrand 
+46736397439

Inspirativ AB | www.inspirativ.se | Smyckegränd 18 12641 Hägersten

closed 2015-06-24 by ekstrand

romangromov, 2015-06-24:
@ekstrand thank you for response, anyway

ekstrand, 2015-06-24:
You are welcome. If wish I hade the time to add that feature, it would be
nice for all of you doing http.
/ Jonas