ekstrand/ESP8266wifi #9: Disable multiple connections

marcfon, 2015-07-11:
I had to make some adjustments to ESP8266wfi get my ESP-01 (0.9.5/0.2.1) to connect successfully to an external server.

line 33: `const char CIPSTART[] PROGMEM = "AT+CIPSTART=\"";`
line 34: `const char CIPCLOSE[] PROGMEM = "AT+CIPCLOSE=";``

add `const char CIPMUX_0[] PROGMEM = "AT+CIPMUX=0";`

and change line 144 to `writeCommand(CIPMUX_0, EOL);`

I don't know why by I get a "link typ ERROR" when communicating over channel 4.

ekstrand, 2015-07-13:
Hi there,

This sounds strange indeed. I’ve just double checked the AT commands ESP8266 AT Instruction Set v0.21 23.01.2015 <http://esp8266.ru/download/esp8266-doc/4A-AT-Espressif%20AT%20Instruction%20Set_v0.21.pdf> for 0.2.1 and 4 is still a valid id when using multiple connection mode  (CIPMUX=1).

Have you tried any other id’s instead of 4, like 0,1,2 or 3? 

If you change CIPMUX=0 it will disable multiple connections, i.e. prevent using esp as a server or connecting to multiple servers at the same time. I’m not sure why that was needed.

Let me know what you find out!

Best regards
Jonas

marcfon, 2015-07-13:
I don't need multiple connections so using CIPMUX=0 is fine for me but I didn't try a different channel actually.

I also had to comment out `_serialOut -> print(channel);` and `writeCommand(COMMA);` in ESP8266wifi::send to create a valid command. But then somehow watchdog() kept thinking I didn't have a valid connection an so I gave up.

marcfon, 2015-07-21:
I managed to get the multi channel setup to work. There was one thing that I finally had to change and that was `flags.connectedToServer = (readCommand(10000, LINKED, ALREADY) > 0);` into `flags.connectedToServer = (readCommand(10000, OK) > 0);` in the `ESP8266wifi::connectToServer()` function.

I think my previous issue was caused by the fact that I was sending `Connection: close` in my get request after which the disconnect method failed.

simonantonio, 2015-11-19:
I've had a similar issue with the MUX issue, I manually send these commands before using the library

    Serial3.begin(9600);
    Serial3.write("AT\r\n");
    delay(100);
    Serial3.write("AT+CIPMUX=1\r\n");
    delay(100);

after this i continue using the library as normal, without that code i get type errors returned from the ESP8266.

(I'm running with an ESP8266-07 against a maple mini)

cederberg, 2015-11-21:
@simonantonio: What if you reduce the sequence to just the following?
```
Serial3.begin(9600);
Serial3.write("AT\r\n");
delay(200);
```
Just adding a delay is strictly speaking wrong. Should also read the serial input until the `OK` response. But if the above works, the single `AT` command can easily be added to the `begin()` sequence.

closed 2016-08-15 by cederberg