ekstrand/ESP8266wifi #13: Errors in SoftwareSerial when combined

TeaZee, 2015-11-06:
A working sketch @115200 with SoftwareSerial reads garbage from the serial stream the moment "#include <ESP8266WiFi.h>" is added to the sketch.

Just including the header file is enough to make the sketch fail to read the characters properly anymore.

cederberg, 2015-11-06:
Probably one of these cause the issue:

```
#include <inttypes.h>
#include <avr/pgmspace.h>
#include "HardwareSerial.h"
```

Please try with them in orde to see which one fails.

TeaZee, 2015-11-06:
Hmm,

#include <inttypes.h>
#include "HardwareSerial.h"

Do not effect the working of the sketch.

#include <avr/pgmspace.h>
Does not compile: "fatal error: avr/pgmspace.h: No such file or directory"

cederberg, 2015-11-07:
Hmm... Could you make a copy of the header and try removing stuff until it unbreaks? I can't see anything serial-related except these three.

TeaZee, 2015-11-07:
Yeah, i'll do some further investigating on what causes the malfunctioning.
I will also try some includes.

Note that the issue is not serial related, since the softserial is "soft" because it doesn't use the UART. It reads and writes on IO pins using interrupts and timing.
The trouble is therefore probably interrupt or maybe timing related. Could be an interrupt clash? A bit odd since it's just the header inclusion that already causes the issues.

I'll investigate further and let you know if it's really ESP8266Wifi or something else.

closed 2016-08-15 by cederberg