ekstrand/ESP8266wifi #21: Serial and swSerial in a

ingenioypsicologia, 2016-02-28:


tuna-f1sh, 2016-03-06:
I swapped them intentionally in favour of hardware serial reliability over software serial. For me, a possible missed char on the debugger is more acceptable then the frustration caused if it happens interfacing the ESP8266. 

A good point though that people may swap between examples and not notice. Could be worth a comment line above highlighting this.

ingenioypsicologia, 2016-03-07:
Ok, I understand now.

lenny1972, 2016-09-15:
Is it possible to use this library with Arduino DUE (using hardware UART port instead using softwareserial)?
If so can you provide me with a working example?
Thanks in advance!

tuna-f1sh, 2016-09-15:
Yes. I don't have a DUE to test with but it should just be a case of replacing _swSerial_ with _Serial_ ([or which serial port you want to use](https://www.arduino.cc/en/Reference/Serial)). See attached.

[ESP8266_tcp_cli_duo.zip](https://github.com/ekstrand/ESP8266wifi/files/474842/ESP8266_tcp_cli_duo.zip)

lenny1972, 2016-09-15:
Thanks for the example.
I'm trying to compile it for Arduino Mega 1280 but the compiler gives me
the following error:

(line 39) 'swSerial' was not declared in this scope'

How is it possible?

lenny1972, 2016-09-15:
Sorry, I didn't read the istruction you gave me (..replacing *swSerial*
 with *Serial..).*

Now it seems tpo compile for Arduino Mega 1280.
As soon I come back home I'll check with the Arduino DUE and let you bknow.

Thanls fopr your prompt reply and support

tuna-f1sh, 2016-09-15:
Re-download the example :). I did a ninja edit on the attached file quickly after posting as it originally contained an unsaved edit.

Wikibear, 2017-02-07:
I get some errors with my wifi. Serial Software 115200 can't be work! I screw down to 9600 and this works fine with my wifi module. Arduino Uno and Nano are don't stable with 115200. Make an UART AT command that this work. You need the debugger script to send AT commands to the chip. Most Firmware versions accept AT commands:

AT+UART(_DEF)=9600,8,1,0,0

Will give a solution.

ekstrand, 2017-02-07:
You don't need to use sw serial.

Wikibear, 2017-02-07:
How to connect to Arduino Nano and read out via serial terminal for debugging then? 
Thanks