ekstrand/ESP8266wifi #23: can this library be used with arduino due

megazero1316, 2016-04-25:
Hello I just want to know if this library can work with arduino due I tried but o think there's something wrong when I modified the example to work with if I am not mistaken I got an error with jmp 0
( don't remember exactly cuz I am away right now from my PC once I get back I will post the error )
 I changed from softserial to serial3 and changed the baud to the required one but when I compile I get an error but when I compile for mega or uno (usng softserial) I get no error

megazero1316, 2016-04-25:
the error :

Build options changed, rebuilding all
C:\Users\nothing\AppData\Local\Temp\ccIRlFiU.s: Assembler messages:

C:\Users\nothing\AppData\Local\Temp\ccIRlFiU.s:76: Error: bad instruction `jmp 0'

exit status 1
Error compiling for board Arduino Due (Programming Port).

`//#include <SoftwareSerial.h>
#include <ESP8266wifi.h>

/* TCP server/client example, that manages client connections, checks for messages
 *  when client is connected and parses commands. Connect to the ESP8266 IP using
 * a TCP client such as telnet, eg: telnet 192.168.0.X 2121
 *  
 *  ESP8266 should be AT firmware based on 1.5 SDK or later
 *
 * 2016 - J.Whittington - engineer.john-whittington.co.uk
 */

//#define sw_serial_rx_pin 4 //  Connect this pin to TX on the esp8266
//#define sw_serial_tx_pin 6 //  Connect this pin to RX on the esp8266
#define esp8266_reset_pin 5 // Connect this pin to CH_PD on the esp8266, not reset. (let reset be unconnected)

#define SERVER_PORT "2121"
#define SSID "SAQ123"
#define PASSWORD "Saq123456"

//SoftwareSerial swSerial(sw_serial_rx_pin, sw_serial_tx_pin);

// the last parameter sets the local echo option for the ESP8266 module..
ESP8266wifi wifi(Serial3, Serial3, esp8266_reset_pin, Serial);

void processCommand(WifiMessage msg);

uint8_t wifi_started = false;

// TCP Commands
const char RST[] PROGMEM = "RST";
const char IDN[] PROGMEM = "*IDN?";

void setup() {

  // start debug serial
  Serial3.begin(115200);
  // start HW serial for ESP8266 (change baud depending on firmware)
  Serial.begin(115200);
  while (!Serial)
    ;
  Serial.println("Starting wifi");

  Serial3.println("Starting wifi");
  //swSerial.write("AT\r\n");
  wifi.setTransportToTCP();// this is also default
  wifi.endSendWithNewline(false); // Will end all transmissions with a newline and carrage return ie println.. default is true

  wifi_started = wifi.begin();
  if (wifi_started) {
    wifi.connectToAP(SSID, PASSWORD);
    wifi.startLocalServer(SERVER_PORT);
  } else {
    // ESP8266 isn't working..
  }
}

void loop() {

  static WifiConnection *connections;

  // check connections if the ESP8266 is there
  if (wifi_started)
    wifi.checkConnections(&connections);

  // check for messages if there is a connection
  for (int i = 0; i < MAX_CONNECTIONS; i++) {
    if (connections[i].connected) {
      // See if there is a message
      WifiMessage msg = wifi.getIncomingMessage();
      // Check message is there
      if (msg.hasData) {
        // process the command
        processCommand(msg);
      }
    }
  }
}

void processCommand(WifiMessage msg) {
  // return buffer
  char espBuf[MSG_BUFFER_MAX];
  // scanf holders
  int set;
  char str[16];

  // Get command and setting
  sscanf(msg.message,"%15s %d",str,&set);
  /* swSerial.print(str);*/
  /* swSerial.println(set);*/

  if ( !strcmp_P(str,IDN) ) {
    wifi.send(msg.channel,"ESP8266wifi Example");
  }
  // Reset system by temp enable watchdog
  else if ( !strcmp_P(str,RST) ) {
    wifi.send(msg.channel,"SYSTEM RESET...");
    // soft reset by reseting PC
    asm volatile ("  jmp 0");
  }
  // Unknown command
  else {
    wifi.send(msg.channel,"ERR");
  }
}`

lenny1972, 2016-09-15:
Did you succeded in making this library working with Arduino DUE? If not, can you suggest any other reliable ESP8266 library that works with Arduino DUE (using hardware serial)?
Thanks

ekstrand, 2016-09-15:
Hi,
Should work if you pass a reference to hw serial instead of sw. Not tested since I don't have an arduino due.

Best of luck!
Regards
Jonas

mrred128, 2017-10-21:
it works fine with Serial1.