ekstrand/ESP8266wifi #31: AT_firmware for ESP_01

MaartenHeuv, 2016-10-17:
Dear Jonas,

Thanks for this very impressive library.

I see you tested with Firmware 0.9.2.4 does this also fit in an ESP_01 ? and where can I obtain this firmware.
I tried flashing AT_firmware V1.3.0.2 for ESP_12 but I'm affraid it is too large.

Thank Maarten

ekstrand, 2016-10-18:
Hi!
I have not tested in any other esp platform or  at firmware version. But I would like to know what configurations that work :)

Regards
Jonas Ekstrand

MaartenHeuv, 2016-10-18:
Dear Jonas,

Thanks for the very quick response.
Sure I hope to validate your work on my platform(s).

Do you know from where to download Firmware 0.9.2.4 ?

Thanks Maarten

MaartenHeuv, 2016-10-18:
Found a firmware quite close https://github.com/espressif/ESP8266_AT

closed 2016-10-20 by MaartenHeuv