ekstrand/ESP8266wifi #33: WIFI_STATE_CONNECT or WIFI_STATE_IDLE

MaartenHeuv, 2016-10-19:
I tried running example "Simple_TCP_Client", however it didn't work.

Looking in the code I found the following:
connectTCP() exits with: setRunningState(WIFI_STATE_CONNECT)
send() test on if(getRunningState() == WIFI_STATE_IDLE)
This is, in my case,  always false so nothing is send. 

I resolved this by changing in:  if(getRunningState() == WIFI_STATE_CONNECT)
Then is works !

I'm totally not sure if my findings or solution is the right !!!!!!!

MaartenHeuv, 2016-10-19:
Jonas,

Not sure, but:
RunningState WIFI_STATE_IDLE comes after the reception of "SEND OK", this comes in my case only after a successfull send() command, however to do a send() first is tested for  WIFI_STATE_IDLE. So first transmission is impossible.

ekstrand, 2016-10-19:
Martin it's been a while since I looked at the code. I have around 5 devices online 24/7 soo I know it worked at one time. I can't be 100% certain about the latest version since I have not upgraded the lib with the latest contributed patches.

If you verify that there is a bug I really welcome a patch :)

Regards
Jonas

MaartenHeuv, 2016-10-20:
Sorry,

My fault: There is no issue / bug

Maarten

MaartenHeuv, 2016-10-20:
There was no issue

closed 2016-10-20 by MaartenHeuv