ekstrand/ESP8266wifi #43: Link error

bbmachado, 2017-03-27:
Hi,

I´m working a project using ESP-8266 ESP-01, but when i try connect do a server return Link Error. When I use the AT Commands direct by serial terminal it´s work fine. Any ideias?

The code:

`#include <ESP8266wifi.h>
#include <SoftwareSerial.h>

#define SSID "My network"
#define PASSWORD "xxxxxxxxxx"
#define sw_serial_rx_pin 3 //  Connect this pin to TX on the esp8266
#define sw_serial_tx_pin 2 //  Connect this pin to RX on the esp8266
#define esp8266_reset_pin 8 // Connect this pin to CH_PD on the esp8266, not reset. (let reset be unconnected)

uint8_t wifi_started = false;
SoftwareSerial swSerial(sw_serial_rx_pin, sw_serial_tx_pin);
// the last parameter sets the local echo option for the ESP8266 module..
ESP8266wifi wifi(swSerial, swSerial, esp8266_reset_pin,Serial);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  swSerial.begin(115200);
  Serial.println("Starting wifi...");

  wifi.begin();
  //wifi.setTransportToTCP();
  wifi.connectToAP(SSID, PASSWORD);
  if (wifi.isStarted()) Serial.println("Just Testing....");
  condicaotempo();
}

void condicaotempo() {
  String location = "/"; // Location where to fetch the new Status
  String cmd = "GET / HTTP/1.1\r\n"; // complete HTTP request
  //cmd += "Host: www.google.com.br\n";

  wifi.connectToServer("www.google.com.br", "80");
  delay(1000);
  wifi.send(SERVER, cmd);

  WifiMessage in = wifi.listenForIncomingMessage(2000);
  if (in.hasData)
  {
    Serial.print("Response: ");
    Serial.println(in.message);
  } else {
    Serial.println("I can´t connect to server :-(");
  }
  wifi.disconnectFromServer();
}

void loop() {
  // put your main code here, to run repeatedly:

}`

Thanks for helping

Fox51, 2017-03-30:
Hi, there

Did you try to set first the command AT+CIPMUX=1 before setting the connectToServer(), you could try something like this : swSerial.write("AT+CIPMUX=1\r\n")

bbmachado, 2017-03-30:
Hi Fox51,

I used this command, but in the begining of the program. I will try again, but before the command connectToServer().

Thanks